VLSI-EDA/PoC #1: Fail when running testbench

marzoul, 2015-09-02:
On operating system GNU/Linux x86-64

my_config.vhdl is configured with the following lines:
```
constant MY_BOARD   : string := "Custom";
constant MY_DEVICE  : string := "xc7vx690t-2ffg1761";
```
my_project.vhdl is configured with the following lines:
```
constant MY_PROJECT_DIR       : string := "/home/adrien/Sandbox/poc-library/myproj/";
constant MY_OPERATING_SYSTEM  : string := "LINUX";
```
PoC-Library was configured with the command ./poc.sh --configure
Tools installed: Vivado 2015.1, GHDL and GTKWave

I tried to run the testbenches with the command: ./testbench.sh --ghdl PoC.arith.addw -v -l -d
An error occured, the terminal messages are the following:

```
[adrien] > ./testbench.sh --ghdl PoC.arith.addw -v -l -d
================================================================================
                    The PoC Library - Testbench Service Tool                    
================================================================================

DEBUG: Reading PoC configuration from '/home/adrien/Sandbox/poc-library/PoC/py/config.private.ini' and '/home/adrien/Sandbox/poc-library/PoC/py/config.public.ini'
DEBUG: Reading testbench configuration from '/home/adrien/Sandbox/poc-library/PoC/tb/configuration.ini'
PoC.arith.addw
  preparing simulation environment...
  Commands to be run:
  1. Change working directory to temporary directory
  2. Parse filelist file.
    a) For every file: Add the VHDL file to GHDL's compile cache.
  3. Compile simulation
  4. Run simulation
  ----------------------------------------
  cd "/home/adrien/Sandbox/poc-library/PoC/temp/ghdl"
DEBUG: Reading filelist '/home/adrien/Sandbox/poc-library/PoC/tb/arith/arith_addw_tb.files'
  running analysis for every vhdl file...

DEBUG: call ghdl: ['/usr/bin/ghdl', '-a', '-P.', '--syn-binding', '--std=93', '--work=poc', '/home/adrien/Sandbox/poc-library/PoC/tb/common/my_config_ML505.vhdl']
    command: /usr/bin/ghdl -a -P. --syn-binding --std=93 --work=poc /home/adrien/Sandbox/poc-library/PoC/tb/common/my_config_ML505.vhdl
ghdl messages for : /home/adrien/Sandbox/poc-library/PoC/tb/common/my_config_ML505.vhdl
--------------------------------------------------------------------------------
/home/adrien/Sandbox/poc-library/PoC/tb/common/my_config_ML505.vhdl:47:14:warning: package "my_config" does not require a body

DEBUG: call ghdl: ['/usr/bin/ghdl', '-a', '-P.', '--syn-binding', '--std=93', '--work=poc', '/home/adrien/Sandbox/poc-library/PoC/tb/common/my_project.vhdl']
    command: /usr/bin/ghdl -a -P. --syn-binding --std=93 --work=poc /home/adrien/Sandbox/poc-library/PoC/tb/common/my_project.vhdl
ghdl messages for : /home/adrien/Sandbox/poc-library/PoC/tb/common/my_project.vhdl
--------------------------------------------------------------------------------
/home/adrien/Sandbox/poc-library/PoC/tb/common/my_project.vhdl:50:14:warning: package "my_project" does not require a body

ERROR: This testbench requires some Xilinx Primitves. Please configure Xilinx ISE / Vivado

Please run 'poc.[sh/cmd] --configure' in PoC root directory.
```

Paebbels, 2015-09-02:
Bug approved.

The testbench script doesn't *fallback* to use the Vivado installation directory, if ISE is not installed on the system. As a result the script can not find the primitives like `MUXCY.vhd` from the unisim library, which are required for the `PoC.arith.addw` testbench.

Paebbels, 2015-09-02:
This commit should fix it.

closed 2015-09-04 by Paebbels