VLSI-EDA/PoC #8: Unable to compile sim_waveform.vhdl

albertschulz, 2016-04-18:
I'm not able to compile the file "sim_waveform.vhdl" using Vivado (Version 2015.2).
I get the following errors:

```
ERROR: [VRFC 10-925] indexed name is not a time [/home/albert/git/dnk7/src/hw/PoC/sim/sim_waveform.vhdl:502]
ERROR: [VRFC 10-925] indexed name is not a time [/home/albert/git/dnk7/src/hw/PoC/sim/sim_waveform.vhdl:531]
ERROR: [VRFC 10-724] found '0' definitions of operator "+", cannot determine exact overloaded matching definition for "+" [/home/albert/git/dnk7/src/hw/PoC/sim/sim_waveform.vhdl:733]
ERROR: [VRFC 10-925] indexed name is not a time [/home/albert/git/dnk7/src/hw/PoC/sim/sim_waveform.vhdl:743]
ERROR: [VRFC 10-724] found '0' definitions of operator "&", cannot determine exact overloaded matching definition for "&" [/home/albert/git/dnk7/src/hw/PoC/sim/sim_waveform.vhdl:745]
ERROR: [VRFC 10-1471] type error near fs ; current type time; expected type real [/home/albert/git/dnk7/src/hw/PoC/sim/sim_waveform.vhdl:748]
ERROR: [VRFC 10-1471] type error near delay ; current type time; expected type real [/home/albert/git/dnk7/src/hw/PoC/sim/sim_waveform.vhdl:889]
ERROR: [VRFC 10-1471] type error near p ; current type time; expected type real [/home/albert/git/dnk7/src/hw/PoC/sim/sim_waveform.vhdl:976]
```

Any ideas?

Paebbels, 2016-04-18:
Hello Albert,

Vivado support is still experimental.

There are many issues in xelab and xsim. We got some reported bugs accepted by Xilinx, but I don't know when their fixes will be available in the production releases.

Xilinx forum: [Compiling a simulation with VHDL-2008 features enabled, results  in falsely generated code for clang](https://forums.xilinx.com/t5/Simulation-and-Verification/Compiling-a-simulation-with-VHDL-2008-features-enabled-results/td-p/689245)

In the meanwhile, can you use other simulators like GHDL (0.34dev is preferred) or QuestaSim/ModelSim?

Regards
    Patrick

Paebbels, 2016-04-18:
Today, Xilinx released Vivado 2016.1, I'll investigate it and keep you updated.