VLSI-EDA/PoC #10: Sphinx Documentation

Paebbels, 2016-05-16:
1. Write the documentation in reST
2. Transform existing documentation from *.md files and wiki pages into reST
3. Add a flow to compile the documentation on ReadTheDocs.org

fer-rum, 2016-05-17:
Okay… Why? 
What is wrong with markdown? 
You are aware that you will most likely break existing document representation by switching the notation language?

Paebbels, 2016-05-17:
Sphinx and reST have many advantages ...

- reST is more powerful than MD
- automatic documentation builds on GitHub push events
- customizable themes and templates
- versioning
- better navigation
- keyword searching and indexing
- building PDF files is possible

And of cause, reST can read/parse and import MD files, while changing the documentation format

fer-rum, 2016-05-17:
Accepted.

Paebbels, 2016-10-28:
A initial version of an auto generated documentation is now available on branch master. The documentation is generated with Sphinx and hosted at ReadTheDocs.io:

http://poc-library.readthedocs.io

Merge commit: f9e431c964f61b96ea97b20205c17eb8a2b52469

closed 2016-10-28 by Paebbels