VLSI-EDA/PoC #19: Issue synthesizing uart_bclk.vhdl

albertschulz, 2016-06-23:
During the synthesis step the uart_bclk is being removed. I'm using Vivado 2016.2

I get the following relevant outputs:

> INFO: [Synth 8-4471] merging register 'blkUART.uart_bclk/bclk_x8_r_reg' into 'blkUART.uart_bclk/bclk_r_reg' [/home/albert/git/dnk7/src/hw/PoC/src/io/uart/uart_bclk.vhdl:104]
> INFO: [Synth 8-3333] propagating constant 0 across sequential element (\blkUART.uart_bclk/bclk_r_reg )
> WARNING: [Synth 8-3332] Sequential element (blkUART.uart_bclk/bclk_r_reg) is unused and will be removed from module kc705_top.

This is how I instantiate the component:
```
		uart_bclk : entity PoC.uart_bclk
		generic map (
			CLOCK_FREQ => 50 MHz
		)
		port map (
			clk 		=> clk_comp,
			rst			=> rst,
			bclk		=> bclk,
			bclk_x8 => bclk_x8
		);
```

The following alternative solution works fine:
```
    bclk_gen_x8: entity PoC.arith_counter_free
      generic map (
        DIVIDER => 50000000/(8*115200)
      )
      port map (
        clk => clk_comp,
        rst => '0',
        inc => '1',
        stb => bclk_x8
      );
    bclk_gen_x1: entity PoC.arith_counter_free
      generic map (
        DIVIDER => 8
      )
      port map (
        clk => clk_comp,
        rst => '0',
        inc => bclk_x8,
        stb => bclk
      );
```

preusser, 2016-06-23:
Vivado has some issues in the support of user-defined physical data types. It is likely that the use of the PoC-defined FREQ type produces improper DIVIDERs that lead to constant outputs so that the component is optimized away.

mzabeltud, 2016-06-23:
This issue is not reproducible on the branch "Vivado" and Vivado 2015.4. Please use this branch when synthesizing with Vivado. I will check it with Vivado 2016.2 soon, but I do not expect a different result.

mzabeltud, 2016-06-23:
Also not reproducible with Vivado 2016.2 and branch "Vivado".

closed 2016-06-23 by mzabeltud