VLSI-EDA/PoC #28: Multiple Examples in the Documentation are the same

befedo, 2016-11-17:
Hello Guys ...

I don't  know if this is a real issue. But while i'm reading the documentation i noticed that there where multiple examples the same.

More precisely the file 'PoC/docs/UsingPoC/Synthesis.rst' contains multiple occurrences of '.\poc.ps1 quartus PoC.arith.prng --board=DE4' from the quartus example.

Thank you for your time and best regards

mzabeltud, 2016-11-17:
Yes, you are right. This is a copy & paste error. Will be fixed soon.

Paebbels, 2016-11-21:
Hello Marc,

the [latest documentation](http://poc-library.readthedocs.io/en/latest/UsingPoC/Synthesis.html) is now fixed. It also contains better document and section linking. The command line switch names are now clickable and direct you to the detailed command line documentation. I also updated the documentation of pre-compile scripts. The updates for the simulation page will be online in a few days.

Thank you for reading our documentation and reporting this issue :)

Kind regards
    Patrick

closed 2016-11-21 by Paebbels

befedo, 2016-11-21:
Thanks Patrick for the quick fix and the corespondending time and work. This is still an amazing project, from now on with an quite better documentation ... ;)

Thanks for your time and best regards,

Marc

Paebbels, 2016-11-21:
May I ask: Are you using PoC in a design or are you just experimenting with our IP cores / tool chain?

I'm also working on a better VHDL reader for a better documentation extraction from VHDL source files. Another new documentation feature is the wavedrom support: See [PoC.io.ddrio.in](http://poc-library.readthedocs.io/en/latest/IPCores/io/ddrio/ddrio_in.html) which has now a waveform rendered in HTML+JS+SVG :)

befedo, 2016-11-21:
I've chosen PoC as an example for documenting and validating my VHDL resources. And of course there where nice implementations in this library I've just started to digging deeper ...

Furthermore I planed to use techniques like OSVVM/VUnit for testing purposes, but there's still a lot for me to learn about. If you got some tutorials as a starting point, it would be nice if you could share them.

Sorry for my bad english, this is not my native language.

Best Regards,

Marc

Paebbels, 2016-11-21:
You have a Gitter message ...