VLSI-EDA/PoC #39: PyInfra: Add tool support for Synopsys Synplify Pro

Paebbels, 2016-12-03:
Add synthesizer support for Synopsys Synplify Pro.

Synplify is also included in Lattice Diamond and can be used as an alternative to the Lattice Synthesis Engine (LSE).

bergnoli, 2020-12-20:
I had troubles using Synplify pro ( Microsemi ) in ocram ( true dual port ram) inference. It seems that Synplify pro does not like the multi clock processes: is that true also for other tools?

Paebbels, 2020-12-20:
`ocram` was tested with lots of tools (ISE, Vivado, Quartus, LSE (Lattice Synthesis Engine, ...).

Please open a new issue, for your problem of how VHDL is written to specify a true dual port RAM. This issue is about Python support in using Simplify Pro.

----------------
In general, the syntax using 2 clocks in one process is correct VHDL and more over this how to write dual port RAMs. This was also specified in [IEEE Std. 1076.6-2004](https://standards.ieee.org/standard/1076_6-2004.html). Normally, Simplify Pro has a very good understanding of VHDL including VHDL-2008 support for synthesis.