VLSI-EDA/PoC #46: News is in Latin on readthedocs front page.

GadgetSteve, 2017-04-12:
I don't think that many will find the [News ](http://poc-library.readthedocs.io/en/release/#news) section enlightening:

> News
> 13.05.2016 - PoC 1.0.0 was released.
> Lorem ipsum dolor sit amet, consetetur sadipscing elitr, sed diam nonumy eirmod tempor invidunt ut labore et dolore magna aliquyam erat, sed diam voluptua. At vero eos et accusam et justo duo dolores et ea rebum. Stet clita kasd gubergren, no sea takimata sanctus est Lorem ipsum dolor sit amet. Lorem ipsum dolor sit amet, consetetur sadipscing elitr, sed diam nonumy eirmod tempor invidunt ut labore et dolore magna aliquyam erat, sed diam voluptua. At vero eos et accusam et justo duo dolores et ea rebum. Stet clita kasd gubergren, no sea takimata sanctus est Lorem ipsum dolor sit amet

mzabeltud, 2017-04-13:
Hi Steve, thanks for pointing that out. I will fix it soon.

mzabeltud, 2017-04-18:
Fixed with commit 11b9ef784713b64072ce8f698a737d5799c10ecd

closed 2017-04-18 by mzabeltud