VLSI-EDA/PoC #48: Error for string.vhdl - Value := resize(slv, Value'length);

bhamadicharef, 2017-08-18:
I want to use the function raw_format_slv_hexfrom string.vhdl from 
VLSI-EDA/PoC under /src/common/strings.vhdl
Vivado 2016.2 gives an error on line 359 about the following

Value := resize(slv, Value'length);

Error message is
[Synth 8-2778] type error near resize ; expected type std_logic_vector

How to fix ? Is it specific to Vivado ?

Brahim @ DCT :: DSI

bhamadicharef, 2017-08-18:
Seems that it gets fixed by 

Value := std_logic_vector(resize(signed(slv), Value'length));

preusser, 2017-08-18:
Thanks for your feedback! Please, use the `unsigned` version version of `resize` to make it semantically correct. Would you want to turn this fix into a pull request then?

Paebbels, 2017-08-18:
The `resize` function is define in [`utils.vhdl`](https://github.com/VLSI-EDA/PoC/blob/master/src/common/utils.vhdl?ts=2#L211). There is no need to rewrite the code.

The question is why doesn't Vivado 2016.2 find that overload.

preusser, 2017-08-18:
`resize` on `std_logic_vector` is ill-defined as it has to make assumptions about signedness. I would rather fence in its use and put it on a deprecation path.

Paebbels, 2017-08-18:
Why is it ill-defined? It makes no assumptions about signedness!

See at the definition in utils, how *you* wrote it. It's using a fill parameter.

preusser, 2017-08-18:
Yes, and it even has implied semantics using the index direction as a hint for alignment. I would now consider this another feature hard to convey. Although I admittedly tried, `std_logic_vector` simply does not have an intuitively right semantics for `resize`. And I do not see why I should not allow my own mistakes to phase out - at least, eventually. The standard libraries provide valid implementations as soon as the types have clear semantics associated with them. Just use those.

Paebbels, 2017-08-18:
Sounds reasonable.

**Usage of `resize`**
* [string.vhdl](https://github.com/VLSI-EDA/PoC/blob/master/src/common/strings.vhdl?ts=2)
  * [line 305](https://github.com/VLSI-EDA/PoC/blob/master/src/common/strings.vhdl?ts=2#L305)
  * [line 359](https://github.com/VLSI-EDA/PoC/blob/master/src/common/strings.vhdl?ts=2#L359)

Where else is that function used?

preusser, 2017-08-20:
Mmhhh, the first of these uses would even be semantically correct ...