VLSI-EDA/PoC #49: configuration crashes at Intel Modelsim setup

jobermueller, 2017-08-31:
Hi,
when configuring PoC for Intel Modelsim Starter Edition, I encountered the following problem:

```
Configuring installed tools
---------------------------
Configuring PoC
  Installation directory: /home/johannes/tries/foss-applications/PoC (found in environment variable)
Configuring Git
  Is Git installed on your system? [Y/n/p]: Y
  Git installation directory [/usr/bin]: /usr/bin
  Install Git mechanisms for PoC developers? [y/N/p]: N
  Git is now configured.
Configuring Aldec
  Are Aldec products installed on your system? [Y/n/p]: n
  Aldec is now configured.
Configuring Altera
  Are Altera products installed on your system? [Y/n/p]: n
Configuring Intel
  Are Intel products installed on your system? [Y/n/p]: Y
  Intel installation directory [/opt/Intel]: /home/johannes/local_programs/intelFPGA_lite
Configuring Intel Quartus
  Is Intel Quartus Prime installed on your system? [Y/n/p]: y
  16.1 version [16.1]: 17.0
  Intel Quartus installation directory [/home/johannes/local_programs/intelFPGA_lite/17.0/quartus]: /home/johannes/local_programs/intelFPGA_lite/17.0/quartus
  Checking Altera Quartus version... (this may take a few seconds)
  Intel Quartus Prime is now configured.
Configuring Intel ModelSim
  Is ModelSim Intel Edition installed on your system? [Y/n/p]: y
FATAL: An unknown or unhandled exception reached the topmost exception handler!
  Exception type:      TypeError
  Exception message:   _ConfigureEdition() takes 1 positional argument but 3 were given
  Caused by:         _ConfigureEdition in file '/home/johannes/tries/foss-applications/PoC/py/ToolChains/Intel/ModelSim.py' at line 113
--------------------------------------------------------------------------------
  File "/home/johannes/tries/foss-applications/PoC/py/PoC.py", line 1247, in main
    poc.Run()
  File "/home/johannes/tries/foss-applications/PoC/py/PoC.py", line 368, in Run
    ArgParseMixin.Run(self)
  File "/home/johannes/tries/foss-applications/PoC/py/lib/pyAttribute/ArgParseAttributes.py", line 181, in Run
    args.func(self, args)
  File "/home/johannes/tries/foss-applications/PoC/py/PoC.py", line 443, in HandleConfiguration
    configurator.ConfigureAll()
  File "/home/johannes/tries/foss-applications/PoC/py/ToolChains/__init__.py", line 775, in ConfigureAll
    self._ConfigureTools(self._configurators)
  File "/home/johannes/tries/foss-applications/PoC/py/ToolChains/__init__.py", line 846, in _ConfigureTools
    self._ConfigurationLoop(configurator)
  File "/home/johannes/tries/foss-applications/PoC/py/ToolChains/__init__.py", line 899, in _ConfigurationLoop
    elif (self._host.Platform == "Linux"):    configurator.ConfigureForLinux()
  File "/home/johannes/tries/foss-applications/PoC/py/ToolChains/__init__.py", line 280, in ConfigureForLinux
    self.ConfigureForAll()
  File "/home/johannes/tries/foss-applications/PoC/py/ToolChains/Intel/ModelSim.py", line 92, in ConfigureForAll
    changed,edition = self._ConfigureEdition()
  File "/home/johannes/tries/foss-applications/PoC/py/ToolChains/Intel/ModelSim.py", line 113, in _ConfigureEdition
    edition =         super()._ConfigureEdition(IntelModelSimEditions, defaultEdition)
--------------------------------------------------------------------------------
Please report this bug at GitHub: https://github.com/VLSI-EDA/PoC/issues
--------------------------------------------------------------------------------
```

I looked for the cause and found the following workaround (same as it's done in the respective files for Altera):
````
diff --git a/py/ToolChains/Intel/ModelSim.py b/py/ToolChains/Intel/ModelSim.py
index b1476f2..bae782a 100644
--- a/py/ToolChains/Intel/ModelSim.py
+++ b/py/ToolChains/Intel/ModelSim.py
@@ -34,7 +34,7 @@ from subprocess                   import check_output
 
 from lib.Functions                import Init
 from ToolChains                   import ConfigurationException, EditionDescription, Edition
-from ToolChains.Mentor.ModelSim   import ModelSimException as Mentor_ModelSimException, ModelSimPEConfiguration as Mentor_ModelSim_Configuration
+from ToolChains.Mentor.ModelSim   import ModelSimException as Mentor_ModelSimException, Configuration as Mentor_ModelSim_Configuration
 from ToolChains.Intel             import IntelException
 
 
@@ -89,14 +89,14 @@ class Configuration(Mentor_ModelSim_Configuration):
 				self.ClearSection()
 			else:
 				# Configure ModelSim version
-				edition = self._ConfigureEdition()
+				changed,edition = self._ConfigureEdition()
 
 
 				configSection = self._host.PoCConfig[self._section]
 				if (edition is IntelModelSimEditions.ModelSimIntelEdition):
 					configSection['InstallationDirectory'] = self._host.PoCConfig.get(self._section, 'InstallationDirectory', raw=True).replace("_ase", "_ae")
 				elif (edition is IntelModelSimEditions.ModelSimIntelStarterEdition):
-					configSection['InstallationDirectory'] = self._host.PoCConfig.get(self._section, 'InstallationDirectory', raw=True).replace("_ase", "_ase")
+					configSection['InstallationDirectory'] = self._host.PoCConfig.get(self._section, 'InstallationDirectory', raw=True).replace("_ae", "_ase")
 
 				self._ConfigureInstallationDirectory()
 				binPath = self._ConfigureBinaryDirectory()
````

This furthermore fixes the default installation path ("ase" instead of "ae" for starter edition of modelsim). If you prefer a pull request over the patch, just let me know.

Paebbels, 2017-08-31:
Hello @jobermueller,

what Git branch have you used?
The ModelSim setup routines are under construction. I have implemented a lot of changes, but these are not uploaded yet, because some testing is needed. I need to change and reorganize a lot in the ModelSim configuration routines due to some new ModelSim flavors like Intel ModelSim Edition and Microsemi ModelSim Edition.

As a work around you can specify a normal ModelSim and specify the path your Altera version. You can also skip ModelSim setup in a first configuration round to get a valid setup and restart configuration for Mentor or Altera tools.

Kind regards
    Patrick

jobermueller, 2017-09-01:
I used the master branch, with the last commit being 3c6e9265fd21305a9c9fe79e561ae2b85fb12eb5.
Regarding the workaround - the patch posted above did it for me, just wanted to inform you of that issue.

Paebbels, 2017-09-01:
I appreciate your work.
Unfortunately, there are a lot of different ModelSim flavors with different feature sets, each with another directory setup.

fer-rum, 2017-09-07:
So the natural conclusion would be to make the directories configurable and provide a meaningful default.

Paebbels, 2017-09-07:
The directories can be configured by the user, but before he can type in the directory, the system is guessing default directories based on vendor, tool name, tool version, tool flavor and operating system. This solution is presented to the user as a default (for pressing [return]).