VLSI-EDA/PoC #58: Submodules are not up-to-date

FranzForstmayr, 2018-07-06:
Some Submodule paths, named pyIPCMI and uvvm failed when cloning recursive.

Paebbels, 2018-07-07:
I I found the problem for UVVM.

The repository clone under github.com/VLSI-EDA is called `UVVM_ALL`, which is the original name. Some time ago, it was renamed to `UVVM` but such an operation at the original repository does not update our cloned repository. I renamed our fork of `UVVM_ALL` to `UVVM`.

FranzForstmayr, 2018-07-07:
The submodules are working now. 
For some reason i see unstaged changes in 'PoC/docs/_themes/sphinx_rtd_theme' submodule, from'docs/make.bat' after a clean 'git clone --recursive'. Can you reproduce this?

FranzForstmayr, 2018-07-08:
It looks like there's a line ending issue, directly in this repository.
https://github.com/VLSI-EDA/sphinx_rtd_theme

Paebbels, 2018-07-08:
Sorry, I cannot reproduce this on my Windows machine. `sphinx_rtd_theme` is a fork with minor modifications.

* What files are listed?
* What Git version are you using?

Even if there are files with line ending issues, this cannot result in unstaged files.

FranzForstmayr, 2018-12-20:
This one should get resolved with this:
https://github.com/rtfd/sphinx_rtd_theme/issues/708