VLSI-EDA/PoC #62: configuration problem

sasamy1, 2018-10-26:
hi everyone.
i faced with pyIPCMI problem. somebody  can help me?
how can i solve this problem?

i referenced POC library document 1.2.0. and exactly followed chap 5.1 to 5.4.1.

[image]

Paebbels, 2018-10-26:
Hi @sasamy1,

I can reproduce this bug and I'll try to push a fix a.s.a.p.

What Python version are you using?

sasamy1, 2018-10-27:
I use python 3.7.1.
Thanks.

lavovaLampa, 2019-05-15:
Any news on the fix? I am hitting the same problem.

oholimoli, 2020-05-11:
I have the same problem :/. Has anybody found a solution?

Peppar, 2020-06-17:
Hello, I was able to fix this specific error by renaming a file in the pyIPCMI submodule, pull request here:
https://github.com/VLSI-EDA/pyIPCMI/pull/1
I'm new to this repo but throwing this out there, sorry if I'm missing something.

poneill1001, 2023-08-29:
I had this issue as well. Per suggestions here, I renamed the file:

 PoC\lib\pyIPCMI\pyIPCMI\ToolChain\PoC.py

To:

PoC\lib\pyIPCMI\pyIPCMI\ToolChain\pyIPCMI.py

Issue was resolved.