VLSI-EDA/PoC #67: Error during configuration

mbieker, 2020-09-14:
Hello,

I tried to setup PoC as described in the Quickstart Guide in the readme. After experiencing the bug described in #62 and applying the suggested fix (renaming PoC.py in the submodule) the following error occured during configuration:
```
FATAL: An unknown or unhandled exception reached the topmost exception handler!
  Exception type:      KeyError
  Exception message:   'INSTALL.Intel.Quartus'
  Caused in:           __getitem__ in file '/usr/lib64/python3.6/configparser.py' at line 959
--------------------------------------------------------------------------------
  File "/home/bcmdev/dev/PoC/lib/pyIPCMI/FrontEnd.py", line 86, in main
    pyIPCMI.Run()
  File "/home/bcmdev/dev/PoC/lib/pyIPCMI/pyIPCMI/__init__.py", line 406, in Run
    ArgParseMixin.Run(self)
  File "/home/bcmdev/dev/PoC/lib/pyIPCMI/lib/pyAttribute/ArgParseAttributes.py", line 187, in Run
    args.func(self, args)
  File "/home/bcmdev/dev/PoC/lib/pyIPCMI/pyIPCMI/__init__.py", line 503, in HandleConfiguration
    configurator.ConfigureAll()
  File "/home/bcmdev/dev/PoC/lib/pyIPCMI/pyIPCMI/ToolChain/__init__.py", line 798, in ConfigureAll
    self._ConfigureTools(self._configurators)
  File "/home/bcmdev/dev/PoC/lib/pyIPCMI/pyIPCMI/ToolChain/__init__.py", line 898, in _ConfigureTools
    self._ConfigureDefaultTools()
  File "/home/bcmdev/dev/PoC/lib/pyIPCMI/pyIPCMI/ToolChain/__init__.py", line 944, in _ConfigureDefaultTools
    selector.Select()
  File "/home/bcmdev/dev/PoC/lib/pyIPCMI/pyIPCMI/ToolChain/Altera/Quartus.py", line 149, in Select
    editions = self._GetConfiguredEditions(QuartusEditions)
  File "/home/bcmdev/dev/PoC/lib/pyIPCMI/pyIPCMI/ToolChain/__init__.py", line 666, in _GetConfiguredEditions
    if (len(self._host.Config[edition.Section]) > 0):
  File "/usr/lib64/python3.6/configparser.py", line 959, in __getitem__
    raise KeyError(key)
--------------------------------------------------------------------------------
Please report this bug at GitHub: https://github.com/VLSI-EDA/pyIPCMI/issues
```

Steps to reproduce:

1. Download PoC via `git clone --recursive`
2. Rename `mv lib/pyIPCMI/pyIPCMI/ToolChain/PoC.py lib/pyIPCMI/pyIPCMI/ToolChain/pyIPCMI.py`
3. Run configuration script:
```
./poc.sh configure
================================================================================
                             pyIPCMI - Service Tool
================================================================================
No private configuration found. Generating an empty pyIPCMI configuration...
Explanation of abbreviations:
  Y - yes      P        - pass (jump to next question)
  N - no       Ctrl + C - abort (no changes are saved)
Upper case or value in '[...]' means default value
--------------------------------------------------------------------------------

Configuring installed tools
---------------------------
Configuring PoC
  Installation directory: /home/bcmdev/dev/PoC/lib/pyIPCMI (found in environment variable)
Configuring Git
  Is Git installed on your system? [Y/n/p]: Y
  Git installation directory [/usr/bin]: /usr/bin
  Install Git mechanisms for pyIPCMI developers? [y/N/p]: N
  Git is now configured.
Configuring Aldec
  Are Aldec products installed on your system? [Y/n/p]: n
  Aldec is now configured.
Configuring Altera
  Are Altera products installed on your system? [Y/n/p]: n
Configuring Intel
  Are Intel products installed on your system? [Y/n/p]: Y
  Intel installation directory [/opt/IntelFPGA]: /opt/quartus
Configuring Intel Quartus Prime
  Is Intel Quartus Prime installed on your system? [Y/n/p]: Y
  17.0 version [17.0]: 18.1
  Intel Quartus Prime installation directory [/opt/quartus/18.1/quartus]: /opt/quartus/18.1/quartus
  Checking Altera Quartus version... (this may take a few seconds)
  Intel Quartus Prime is now configured.
Configuring Intel Quartus Prime Lite
  Is Intel Quartus Prime installed on your system? [Y/n/p]: n
Configuring Intel ModelSim
  Is ModelSim Intel Edition installed on your system? [Y/n/p]: n
Configuring Intel ModelSim (Starter Edition)
  Is ModelSim Intel Edition installed on your system? [Y/n/p]: n
Configuring Lattice
  Are Lattice products installed on your system? [Y/n/p]: n
Configuring Mentor
  Are Mentor products installed on your system? [Y/n/p]: n
Configuring Xilinx
  Are Xilinx products installed on your system? [Y/n/p]: n
Configuring GHDL
  Is GHDL installed on your system? [Y/n/p]: n

Configure default tools? [Y/n]: Y
Choosing default tools
```
Any help would be greatly appreciated.

broddo, 2022-01-26:
I know this issue has been open for a while now, but I just thought I'd mention that I also have experienced this issue.

Paebbels, 2022-01-26:
The current plan to solve issues of PoC's Python scripting is as follows:
* PoC's Python scripting got extracted into [pyIPCMI](https://github.com/Paebbels/pyIPCMI) (long time ago)  
 The idea was to make the Python scripting reusable for other IP core libraries or collections. While doing so, some things broke, because parts of pyIPCMI were not properly decoupled.
* pyIPCMI is mostly monolithic with lots of internal knowledge, but not useable by other frameworks.  
  E.g. it has features since 2016, that other competing frameworks have managed to implement in the last 6 years.
* pyIPCMI is currently restructured, decomposed, abstracted and developed from scratch as the foundation of [EDA²](https://edaa-org.github.io/).
  * We defined 10 layers in the EDA² stack.
  * Each layer can be used independently.
  * Each layer is shipped with documentation, usecases/examples, automated unit testing, code coverage analysis, ...
  * All interfaces between layers are well defined and documented.
* The following layers are published
  * pyAttributes
  * pyTooling
  * pyTooling.TerminalUI
  * pyTooling.CLIAbstraction
  * pyEDAA.CLITool
  * pyEDAA.ToolSetup (partially done)
  * pyEDAA.ProjectModel
  * pyEDAA.UCIS
  * pyVHDLModel
  * pySVModel
* These layers are planned or under development:
  * pyEDAA.Workflow
  * pyEDAA.Reports
  * pyEDAA.OutputFilter

The work has started 4 months ago and will take another 3 months.  
File formats of PoC / pyIPCMI might be reviewed and exchanged to more modern/powerful formats like YAML instead of INI.

A branding strategy was developed and we're in contact to other frameworks, who want to reuse layers, contribute features or cooperate with us. On top, projects like OSVB are investigating prototypes and new ideas based on layers provided by EDA².

An overall health status can be found here: https://edaa-org.github.io/Status.html  
Development investments by me into EDA² and pyTooling drastically increased in the last 4 months:  
[image]

Please stay tuned for updates on that topic within the next 3 months.

----------------
/cc @umarcor