VLSI-EDA/PoC #70: Failed to configure

albydnc, 2022-11-01:
I am trying to configure PoC for GHDL under Fedora 35.
When I run the sh script I get:
```
$ ~/PoC/poc.sh configure GHDL
IMPORT ERROR: One or more Python packages are not available in your environment.
Missing package: 'collections'

Run: 'python3 -m pip install -r requirements.txt'
```

I am running on Python 3.10.7 and it has `collections`.

stefanunrein, 2022-11-07:
I have no in-depth knowledge about the scripting part. @Paebbels can you help with this problem?

kbc-mh, 2022-11-24:
I ran into the same problem today. 
I tried it with Python 3.11.0 / 3.9.12 / 3.7.9, all in a virtual env.
I have not found the source of the problem yet. Any help would be highly appreciated.

poneill1001, 2023-08-29:
I had the same issue and was able to fix it by changing the line 41 in PoC\lib\pyIPCMI\lib\ExtendedConfigParser\__init__.py:

from collections  import OrderedDict as _default_dict, ChainMap as _ChainMap, MutableMapping

To:

from collections  import OrderedDict as _default_dict, ChainMap as _ChainMap
from collections.abc  import MutableMapping

Apparently after Python 3.5 MutableMapping was deprecated in collections and relocated to the collections.abc package.  I do not know the full extent of behavior across all the versions to suggest most appropriate fix for the code base.