VLSI-EDA/PoC #74: Check code quality in a continuous way

racodond, 2024-09-20:
Hi,

I analyzed the VHDL code from your PoC project with Linty: https://oss.linty-services.com/dashboard?id=poc&codeScope=overall

Do not get scared by the number of issues :-)
It's just to provide you with data to browse. I activated all the rules, some are mutually opposed, others are not properly configured with your requirements (such as naming conventions).

We offer Linty for free for open-source projects. I'd be glad to organize a call if you're interested in checking code quality in a continuous way through GitHub actions.
We also offer our VS Code extension for free: https://hdl-designer.linty-services.com/

I'm looking forward to hearing from you.

Thank you

Have a good day

David Racodon
CTO at Linty Services

Paebbels, 2024-09-21:
@racodond I checked the results here and there ...

Who ever configured the rules that linty uses, needs to first understand and learn VHDL properly before creating rule-set that are now globally used. Some rules try to enforce coding styles that are not valid any more! Especially when you used VHDL-200* or newer!

Some of the rules even try to enforce bad coding style due to vendor tools not following the VHDL standard. These rules have been commonly established because of tool bugs and problems, but in any decent programming language, your team lead would fire you if you apply such rules to C++ or Java or Python programs.

Please revise the used rules ...

Paebbels, 2024-09-21:
@racodond to give you just TWO examples:

| Ports | Variables |
| --- | --- |
| [image] | [image]
| An entity is not a method or function! You already need 2 port for `Clock` and `Reset`. What can an entity do with just 4 remaining ports? This is a hard | Variables (besides constants and files) are the only objects that can be created in a function. Writing non-trivial functions without variables is IMPOSSIBLE !! The justification, variables lead to synthesis problems has a) not understood VHDL, b) should stop using VHDL and creating hardware and c) think about abandoning signals too, because the problem is not variable specific !! |

racodond, 2024-09-22:
Hi Patrick,

Thanks for having a quick look at Linty and providing your feedback!

> Who ever configured the rules that linty uses, needs to first understand and learn VHDL properly before creating rule-set that are now globally used. Some rules try to enforce coding styles that are [...]

This is not the role of Linty to enforce any coding guidelines.
Defining coding guidelines is the role of design teams. They should define their guidelines according to the context and the constraints of their projects.
Coding guidelines may drastically differ from one project to another. Rules won't be the same on a 10-year-old legacy project or on a new project being able to use the latest VHDL language features. High dependencies on tooling used for synthesis can lead to different guidelines, etc.
The role of Linty is to provide the tooling to automatically enforce coding guidelines defined by design teams. That's why there are rules that are valid for old versions of VHDL but are not for newest versions, rules that are mutually opposed, etc.

> An entity is not a method or function! You already need 2 port for Clock and Reset. What can an entity do with just 4 remaining ports? This is a hard

6, 10, 20 are all good thresholds depending on your context. Again, it is the responsibility of the design team to set the right threshold according to their context.

There is no good default threshold. We went for a very low threshold (as you stated, it's almost impossible to go below 6 ports) because we want designers to think about the right threshold when they activate this rule.

If we went for a high default value (for instance 20) and if the design team didn't customize the threshold according to their context, they would only get notifications when their entities go over 20 ports. It could be quite late in their context. It would have maybe been better to get notified at 12 to start refactoring more easily then. That's why we went for a very low threshold. I was even personally for a default threshold of 1 :-)

> Variables (besides constants and files) are the only objects that can be created in a function. Writing non-trivial functions without variables is IMPOSSIBLE !! The justification, variables lead to [...]

This rule does not apply to your lib for sure and you would not have activated this rule if you've been asked for.

Again, it is context dependent. Some teams want to follow this rule and I guess they have good reasons to in their context (I'm not here and Linty is not here to judge them).

It is defined in this standard for instance: https://github.com/VHDLTool/VHDL_Handbook_CNE/releases/download/CNE_V2.1/handbook_CNE_Edition_V2.1.pdf (STD_06300).

Designers often disagree on coding guidelines. We let them do and offer them the ability to activate and configure rules on what they decide is best for them.

I hope it clarifies things.

This being said, I think we should have activated a lot fewer rules for the "demo" and stick to rules that most designers agree on.

I'd be glad to grant you permissions to update the set of rules on your project. Then you can run analyses by yourself to see what you get with your coding guidelines.

I'd be also glad to organize a video conf to discuss your needs and Linty.

Let me know what you think.

Thank you

Have a good evening

David

racodond, 2024-10-15:
Hi Patrick,

Would you like me to set up a quick call for a discussion around Linty?
Or shall I close the project on our cloud platform: https://oss.linty-services.com/dashboard?id=poc&codeScope=overall ?

Thank you

Have a good evening

David